Argmaster/pygerber #12: Trouble getting started with 2D API

jimmyswimmy, 2022-10-06:
Hi,

I've installed both the pip repository version of pygerber (1.0.0) as well as the git version (1.1.0) and when trying to run the first example command for the 2D API I get the same error:

```python
>>> from pygerber import pillow
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'pillow' from 'pygerber' (/home/j/.conda/envs/gerbers/lib/python3.9/site-packages/pygerber/__init__.py)
```

Again, I get this error whether I install pygerber as pip install pygerber, or if I get the latest github work from...
```bash
conda create -n "gerbers" python=3.9
conda activate gerbers
pip install git+https://github.com/Argmaster/pygerber.git
```

In case you think it could be my install, I've explicitly created a python 3.9 environment (it didn't want to install in 3.10), and have the following packages installed:

```bash
(gerbers) [j@DCAM03 MK18_ShortStave]$ pip freeze
asttokens==2.0.8
backcall==0.2.0
certifi @ file:///croot/certifi_1665076670883/work/certifi
charset-normalizer==2.1.1
click==8.1.3
commonmark==0.9.1
decorator==5.1.1
executing==1.1.0
idna==3.4
ipython==8.5.0
jedi==0.18.1
matplotlib-inline==0.1.6
numpy==1.23.3
packaging==21.3
parso==0.8.3
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.2.0
prompt-toolkit==3.0.31
ptyprocess==0.7.0
pure-eval==0.2.2
pydantic==1.10.2
pygerber @ git+https://github.com/Argmaster/pygerber.git@4761a5aa60ff1d11512fb44aabd103246d9a3019
Pygments==2.13.0
pyparsing==3.0.9
PyR3==0.3.0
PyYAML==6.0
requests==2.28.1
rich==12.6.0
six==1.16.0
stack-data==0.5.1
toml==0.10.2
traitlets==5.4.0
typing_extensions==4.3.0
urllib3==1.26.12
wcwidth==0.2.5
```

Any ideas on what to do? This feels like a 'me error' but usually I can figure those out pretty quickly.

Argmaster, 2022-10-06:
Indeed it is a mistake in docs, try:

```python
from pygerber import API2D

rendered_image = API2D.render_file("docs/gerber/example_gerber.grb")
rendered_image.show()
```

I will fix it, thanks for pointing it out.

arcivanov, 2022-12-13:
Just curious, is the project still maintained?

Argmaster, 2022-12-13:
Not really, no interest was show and I became quite too busy to work on it.

Argmaster, 2022-12-14:
Sure. If any interest will be shown in the future, I may change my mind.

Argmaster, 2023-09-01:
Hi, I wanted to share an update about how to render 2D images with PyGerber 2.0.0
Here is an example code:
```py linenums="1" title="render_file.py"
from pygerber.gerberx3.api import (
      ColorScheme,
      Rasterized2DLayer,
      Rasterized2DLayerParams,
)

# Path to Gerber source file.
source_path = "main_cu.grb"

Rasterized2DLayer(
      options=Rasterized2DLayerParams(
            source_path=source_path,
            colors=ColorScheme.COPPER_ALPHA,
      ),
).render().save("output.png")
```

closed 2023-09-01 by Argmaster