Argmaster/pygerber #13: Is there a future for this project? Should there be future?

Argmaster, 2023-02-13:
Main purpose of this code is to parse gerber (v3) format and create image or 3D model showing content of the file. However, project was abandoned due to lack of time. It is worth mentioning that there are plenty of GUI/web based tools allowing for such visualization. I do not need such software myself and see no reason to use my time to support something that is not used by anyone.

Argmaster, 2023-06-26:
I haven't decided to abandon this project, and quite likely i will never do so, just it won't receive any updates until they are requested. To be honest probably now is the best moment to revive it, as i'll have more free time for next 3 months.

ajw287, 2023-06-26:
Hello!  I only  forked it this morning to have a play - so I wouldn't take anything I say too seriously - I've got the code to open the demo gerbers but not outputs from KiCAD at the moment (probably because I'm using python 3.11 not 3.9).  I want to use your API to graphically 'diff' gerber files  (I use git with my kicad projects but interpreting diffs of the raw gerber text is a bit annoying - although there's a free viewer in kiCAD the features are limited.  I could see using your code to augment git - since the gerbers are text they do well in VC, but the diffs are limited...  this might find applications in things like KiCAD CI (e.g. https://sschueller.github.io/posts/ci-cd-with-kicad-and-gitlab/) which is further using principles of devOps in hardware design.  I also keep meaning to make something to drop the git hash number onto the gerber files (I've done this the past by search and replace in - but it would make sense for this to be done somewhere here).  Also, are you aware that KiCAD is scriptable in python, this means that your code could be useful even if the gerbers aren't kept in the VC (as you could script generating gerbers and checking them..)

closed 2023-08-02 by Argmaster