Argmaster/pygerber #15: pygerber parsing gerber files from kicad

ajw287, 2023-06-29:
Hi,

  I wanted to get some background on this issue.  At the moment I get alot of bad tokens when I use gerbers from sources other than the demo gerber files in the package.

e.g.
$ python3 -m pygerber --pillow --toml "./specfile.yaml" -s "render.png"
Rendering /home/user/Documents/KiCAD/test_gerbers/specfile.yaml as TOML
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/pygerber/__main__.py", line 18, in <module>
    main()
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/pygerber/__main__.py", line 14, in main
    handle_pygerber_cli(sys.argv[1:])
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/pygerber/cli.py", line 92, in handle_pygerber_cli
    handle_pillow_cli(args)
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/pygerber/parser/pillow/cli.py", line 20, in handle_pillow_cli
    image = render_from_toml(args.specfile["filepath"])
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/pygerber/API2D.py", line 53, in render_from_toml
    return PillowProjectSpec.from_toml(file_path).render()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/pygerber/parser/project_spec.py", line 56, in from_toml
    spec = toml.load(file)
           ^^^^^^^^^^^^^^^
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/toml/decoder.py", line 156, in load
    return loads(f.read(), _dict, decoder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/software/gerber-venv/lib/python3.11/site-packages/toml/decoder.py", line 260, in loads
    raise TomlDecodeError("Found invalid character in key name: '" +
toml.decoder.TomlDecodeError: Found invalid character in key name: ':'. Try quoting the key name. (line 1 column 4 char 3)

[test_gerbers.tar.gz](https://github.com/Argmaster/pygerber/files/11904709/test_gerbers.tar.gz)

This is a basic 2 layer gerber file that has most of the basic components of a PCB (see image).  I get similar token issues when using the API,  I've played with deleteing much of the file, but need to get to grips with what is and isn't implemented.  I was wondering if you have any advice on where to start and whether this is something caused by using 3.11 not 3.9?

ajw287, 2023-06-29:
P.S.  I used pygerber to make a start on a gerber graphical diff program.  [Check it out](https://github.com/ajw287/diffgerber)

Argmaster, 2023-07-01:
Thanks for samples, I will check them out soon. 
As about the error you are seeing, you are using YAML specfile, but passing it with `--toml` flag, which causes pygerber to think its TOML file. Use `--yaml` instead and it should work.
I started working on next release (2.0.0) which involes major rework of parsing system. I want to use pyparsing for it, as it's very powerfull tool. 
Meanwhile I have checked your repo and it looks promising. Im not really a PCB guy, thus I'm not  really that hyped.

Argmaster, 2023-07-01:
If you would find time to work on Github Actions configuration, it would be great, as it is quite consuming and I would prefer to concentrate on developing next release. (branch `revolution/2.0.0`)

ajw287, 2023-07-06:
Amazing !  I look forwards to version 2.0.0!

  No worries about the diff tool - its just a toy really - but I would use it and it doesn't exist in this common packages like gerbv.

ajw287, 2023-07-06:
I'll take a look at github actions configuration - I've never used it though.

closed 2023-08-19 by Argmaster