Argmaster/pygerber #25: Progress with pygerber

ajw287, 2023-07-24:
I just wanted to check in  (is there a better way to do this?)  I see you are making lots of progress, let me know when I should start experimenting with pygerber 2.0!

I'm having trouble aligning the images when there are parts of a layer that go NW of the origin,  in this case the images are generated correctly, but my program shows them misaligned?  Is there a way to get the origin out of pygerber?  Perhaps I could start writing some docs as I go along here?

Also, I was wondering if you had any good options for references on the gerber standard?

ajw287, 2023-07-26:
I guess the standard - I am going to have to find a work around for the bounding box issue - is the datum in the gerber files (I could parse them for that information in diffgerber...)?  I can't really show multiple layers if they aren't aligned so it is pretty important.

Argmaster, 2023-07-26:
I'm quite close to releasing 2.0.0-rc1 (but I'm still missing public API, which may turn out to be problematic), so It might turn out that no workarounds will be needed. I will provide absolute bounding box coordinates and coordinate offset used to reposition (min_x, min_y) to 1'st quadrant. Those information should be sufficient to determine positions of images for grb files which were using common coordinate origin (ie. Their coordinate axes are aligned). But I either need help or at least one more week.

ajw287, 2023-07-27:
This will be great!  Don't worry, its meant to be fun!  The bounding box coords in 2.0 would be absolutely amazing.

Argmaster, 2023-07-27:
Hi, I have added some [public API](https://github.com/Argmaster/pygerber/tree/revolution/2.0.0/src/pygerber/gerberx3/api) for 2.0.0, I'd be glad to hear your thoughts on it. 
It was not tested yet, there are no docs on how to use it, only thing I can promise is that I will do my best to add those.

ajw287, 2023-08-02:
What a lot of work!   I probably can't comment here. I hardly understand what you've done!  I can see that from what you've done, it looks like your thinking on the code has evolved, but from a quick peak at the code, I can't work out how to use it! :8

  I have a feeling that you understand it all a lot better than me - so please don't be put off I defer to your knowledge - I'm just playing catch up.   Is there somewhere in the project where I could look at some examples of API usage?  From there I can start experimenting, or creating some tests...

  How do you see the color schemes working?  PCBs are commonly from 1-8 layers but can be alot more (e.g. jlcpcb do up to 20 on a 'standard pool' service) .  I don't think that there are any defined standards for layer/colors - though I know that common layout programs like altium and kicad both use red for front and blue for back (by default) Personally I always find them a bit ugly, but gave up changing them as it just gets confusing.   I think gerbv, just has a list different colors that it uses for successive layers - again I think that you can set them if you want but it is effort.

ajw287, 2023-08-02:
Ah.. ok looking a bit further, I think that the architecture is good (I'm not a software engineer though!)  but I think that its looking better than I could do.  Am I right in thinking that the implementations of classes llike layer are still WIP? Apologies if I've missed something - I can see that you've done a shedload of work on this.

Argmaster, 2023-08-02:
So basically a Layer is a single Gerber file, not necessarily a layer of PCB, ie. silk will be one layer, paste mask will be another, copper top will be next etc. I want to leave it as open and general as possible as I have little experience and I dont want to impose stupid constraints on users. See [usage](https://argmaster.github.io/pygerber/#usage), for now it's only place which explains how it works.

Argmaster, 2023-08-02:
I have opened [Discussions](https://github.com/Argmaster/pygerber/discussions) section which should be a better place for discussions like this one. I have no idea how to use it though, I hope it turns out in the process that it's easy.
I will also close this issue as it has no real use.

closed 2023-08-02 by Argmaster