Argmaster/pygerber #34: Image from Rasterized2DLayer.render/Layer

ajw287, 2023-08-27:
I've finally got around to setting up your v2.0 code - it looks really good!  Thanks for your updated documents - I know its not exciting, but they really helped me get going. 

  So I'm planning to sunset the multi-backend version that I've been developing of diffgerber and concentrate on using pygerber 2.0 (it will probably cut the code length by ~90%!)

   I can get the demo to work - by rendering and saving to file :/ but for my purposes it would be better to just use an Image object in the GUI (since both tk and pygerber use pillow).  I can't seem to work out how to get an image out of the renderResult object though.  What's the best way to approach this when interacting with the API?

ajw287, 2023-08-29:
P.S.  I am apparently giving a talk about [diffgerber](https://github.com/ajw287/diffgerber) at KiCON '23!  (Nobody confirmed with me!  ... I just saw it in the schedule!).  Is there anything that you want said about pygerber?  I am planning on talking about how I added three back ends and struggled with each, then pygerber 2.0 came out, so I'm looking to use that exclusively.

Argmaster, 2023-08-31:
> I've finally got around to setting up your v2.0 code - it looks really good! Thanks for your updated documents - I know its not exciting, but they really helped me to get going.

I'm happy that it was readable enough.

> So I'm planning to sunset the multi-backend version that I've been developing of diffgerber and concentrate on using pygerber 2.0 (it reduced the import code length by ~90%!) and will let me build an application that caters to the API properly without lots of redevelopment.
> 
> For my purposes it would be better to just use an Image object in the GUI (since both tk and pygerber use pillow). I got this working by using:
> 
> ```
>             out =  Rasterized2DLayer(
>                 options=Rasterized2DLayerParams(
>                         dpi=300,
>                         source_path=file_path,
>                         colors=cunning_scheme,
>                 ),
>             )
>             layerImage =  out.render()._result_handle.result
> ```
> 
> This code seems to work, but the images are flipped vertically. It's not a problem as I can flip it in my code, but it implies that I'm probably missing something.

Images are flipped vertically because Pillow (0,0) point is top left corner of image, therefore point `(100, 0)` will be below `(200, 0)`. Internally PyGerber is also flipping them.

What you could do is saving image into BufferIO, with following code like its done [in this test](https://github.com/Argmaster/pygerber/blob/release/2.0.0/test/examples/render_copper_from_path_into_buffer.py).

I will add get_image() method to `RenderingResult` in 2.0.0rc3

Argmaster, 2023-08-31:
> P.S. I am apparently giving a talk about [diffgerber](https://github.com/ajw287/diffgerber) at KiCON '23! (I'd applied a while ago, but nobody confirmed with me! ... I just saw it in the schedule!). Is there anything that you want said about pygerber? I am planning on talking about how I added three back ends and struggled with each, then pygerber 2.0 came out, so I'm looking to use that exclusively.

I don't have anything particular in mid, but it would be great if you would mention it there.

> Also, I might not have installed pygerber properly. When running your demo code I had to change the imports to:
> 
> ```
> from pygerber.gerberx3.api import ColorScheme
> from pygerber.gerberx3.api._layers import (
>       Rasterized2DLayer,
>       Rasterized2DLayerParams,
> )
> ```
> 
> I don't know why.

Please share traceback and `pip show pygerber`.

ajw287, 2023-08-31:
> Please share error traceback and pip show pygerber.

Traceback: 
```
$ python
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pygerber.gerberx3.api.color_scheme import ColorScheme
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pygerber.gerberx3.api.color_scheme'
>>> from pygerber.gerberx3.api.layers import (
...       Rasterized2DLayer,
...       Rasterized2DLayerParams,
... )
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pygerber.gerberx3.api.layers'
>>>
```
pip show:
```
$ pip show pygerber
Name: pygerber
Version: 2.0.0rc2
Summary: Parsing, formatting and rendering toolkit for Gerber X3 file format
Home-page: 
Author: Krzysztof Wisniewski
Author-email: argmaster.world@gmail.com
License: MIT
Location: /home/user/software/diff2-venv/lib/python3.10/site-packages
Requires: click, numpy, pillow, pydantic, pyparsing, tzlocal
Required-by:
```

Argmaster, 2023-08-31:
Ok, examples in README are outdated, here is updated version:

```python
from pygerber.gerberx3.api import (
      ColorScheme,
      Rasterized2DLayer,
      Rasterized2DLayerParams,
)

# Path to Gerber source file.
source_path = "main_cu.grb"

Rasterized2DLayer(
      options=Rasterized2DLayerParams(
            source_path=source_path,
            colors=ColorScheme.COPPER_ALPHA,
      ),
).render().save("output.png")
```

closed 2023-09-03 by Argmaster