Argmaster/pygerber #36: running example: No module named 'pygerber.gerberx3.api.color_scheme'

c2h2, 2023-08-31:
Thanks for the project, just starred.

copied code from README

```
➜ pip3 install pip3 install pygerber
Defaulting to user installation because normal site-packages is not writeable
Collecting pygerber
  Downloading pygerber-1.0.0-py3-none-any.whl (55 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 55.3/55.3 KB 542.6 kB/s eta 0:00:00
Collecting PyR3>=0.2.2
  Downloading PyR3-0.2.3-py3-none-any.whl (23 kB)
Collecting toml>=0.10.0
  Downloading toml-0.10.2-py2.py3-none-any.whl (16 kB)
Requirement already satisfied: numpy>=1.21.0 in ./.local/lib/python3.10/site-packages (from pygerber) (1.25.0)
Collecting pillow>=8.3.0
  Downloading Pillow-10.0.0-cp310-cp310-manylinux_2_28_x86_64.whl (3.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.4/3.4 MB 10.4 MB/s eta 0:00:00
Requirement already satisfied: PyYAML>=5.4.0 in /usr/lib/python3/dist-packages (from pygerber) (5.4.1)
Collecting requests>=2.26.0
  Downloading requests-2.31.0-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.6/62.6 KB 3.1 MB/s eta 0:00:00
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/lib/python3/dist-packages (from requests>=2.26.0->PyR3>=0.2.2->pygerber) (1.26.5)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests>=2.26.0->PyR3>=0.2.2->pygerber) (3.3)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests>=2.26.0->PyR3>=0.2.2->pygerber) (2020.6.20)
Collecting charset-normalizer<4,>=2
  Downloading charset_normalizer-3.2.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (201 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 201.8/201.8 KB 9.3 MB/s eta 0:00:00
Installing collected packages: toml, pillow, charset-normalizer, requests, PyR3, pygerber
  WARNING: The script normalizer is installed in '/home/c2h2/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script pygerber is installed in '/home/c2h2/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed PyR3-0.2.3 charset-normalizer-3.2.0 pillow-10.0.0 pygerber-1.0.0 requests-2.31.0 toml-0.10.2

➜  gits python3 gerber.py 
Traceback (most recent call last):
  File "/Users/c2h2/gits/gerber.py", line 1, in <module>
    from pygerber.gerberx3.api.color_scheme import ColorScheme
ModuleNotFoundError: No module named 'pygerber.gerberx3.api.color_scheme'
```

Any reasons?

c2h2, 2023-08-31:
pip3 install was correct, just copy pasted extra words.

Argmaster, 2023-08-31:
Please try installing latest release candidate:
```
pip install pygerber==2.0.0rc2
```

Also, if possible, share content of your script. What may be happening is that you are using code suited for PyGerber 2.0.0 with PyGerber 1.0.0, and they are not compatible.

ajw287, 2023-08-31:
I used the following:

```
from pygerber.gerberx3.api import ColorScheme
from pygerber.gerberx3.api._layers import (
      Rasterized2DLayer,
      Rasterized2DLayerParams,
)
```

This works for me when running the demo code.

Argmaster, 2023-08-31:
Please refer to #34 with problems with imports `from pygerber.gerberx3.api`, README.md was outdated, I have fixed it and uploaded updated documentation.

c2h2, 2023-09-01:
Hi thanks for the update, seems It installed the  new version. code is below:

```
➜  gits pip3 list | grep pygerber
pygerber           2.0.0rc2

➜  gits python3 gerber2.py       
Traceback (most recent call last):
  File "/Users/c2h2/gits/gerber2.py", line 1, in <module>
    from pygerber.gerberx3.api.color_scheme import ColorScheme
ModuleNotFoundError: No module named 'pygerber.gerberx3.api.color_scheme'
```
code:
```
import pygerber
from pygerber.gerberx3.api.color_scheme import ColorScheme
from pygerber.gerberx3.api.layers import (
      Rasterized2DLayer,
      Rasterized2DLayerParams,
)

# Path to Gerber source file.
source_path = "main_cu.grb"

Rasterized2DLayer(
      options=Rasterized2DLayerParams(
            source_path=source_path,
            colors=ColorScheme.COPPER_ALPHA,
      ),
).render().save("output.png")
```

But i did sucessfully run the test/example files.

Argmaster, 2023-09-01:
TL;DR; issue was caused by mistake in README, already fixed on `main` branch.
Example should have looked like this:

```py linenums="1" title="render_file.py"
from pygerber.gerberx3.api import (
      ColorScheme,
      Rasterized2DLayer,
      Rasterized2DLayerParams,
)

# Path to Gerber source file.
source_path = "main_cu.grb"

Rasterized2DLayer(
      options=Rasterized2DLayerParams(
            source_path=source_path,
            colors=ColorScheme.COPPER_ALPHA,
      ),
).render().save("output.png")
```

closed 2023-09-01 by Argmaster

c2h2, 2023-09-05:
thanks, it is working!