Argmaster/pygerber #37: Difference in rendering between KiCAD gerber viewer and pygerber

ajw287, 2023-08-31:
I noticed this difference in rendering between the pygerber and the same gerber viewed in KiCAD gerber viewer.  I don't know if the KiCAD gerber viewer is specifically removing these artifacts or something, but I thought I'd draw your attention to it as PCB designers look out for these kinds of artefacts (and remove them) as they can be caused during layout .

[image]
[image]

Argmaster, 2023-08-31:
IMO those artifacts are caused by low DPI and therefore low precision. You are comparing low DPI rasterized image to high DPI vector image. Try rendering with DPI like 1000 or 2000.

ajw287, 2023-08-31:
The first one was rendered at 1400, I'm pretty sure artefacts are real, but it would be useful if you could reproduce this.  It is the original demo gerber for pygerber.  The second one is at 400dpi, but the keepouts around the tracks are very different sizes.

ajw287, 2023-09-01:
P.S.  BTW, I just wanted to say that what you have done if F8ciking awesome.   Honestly, in such a short time a whole API that renders the gerber format is awesome!  There is a lack of tools around this technology so,  I'm really stoked that you are developing. :)

Argmaster, 2023-09-01:
I will try to reproduce it, but possible fixes will be included in PyGerber 2.1.0, as we have to release 2.0.0 version soon.

ajw287, 2023-09-06:
Hey,  Just to say no rush - pygerber is already doing great things!  I will show the current setup off at KiCon this weekend - I mention pygerber 1.0 and 2.0 and your name - I hope that's ok.  I have realized that the issue with rendering at least two things (some of which I think you will be aware of) - in the second example the pads disappear,  I think that the rounded rectangles aren't implemented yet (or subroutines?)  I also get lots of warnings in the command line: 

`WARNING:root:Created region with no boundaries.`

Searching the code for this warning, I see that it is when it hits 'G37', e.g. :
`G37*
%TD.AperFunction*%
%TA.AperFunction,Conductor*%`

AperFunction is defined elsewhere in the gerber file.  So I think that this might be because we haven't got to subroutines or something?

I'll check a second example into the diffgerber code.  Hope that you are good.

Argmaster, 2023-09-08:
> I have realized that the issue with rendering at least two things (some of which I think you will be aware of) - in the second example the pads disappear, I think that the rounded rectangles aren't implemented yet (or subroutines?)

Indeed there are at least two separate problems. I will create separate issues for each of them. Problem visible on first set of images probably has to do with sizing of apertures and position rounding, while the second may consist of region boundary drawing. However I have to check first how it interacts with DPI changes.

> I also get lots of warnings in the command line:
>
> `WARNING:root:Created region with no boundaries.`
>
> Searching the code for this warning, I see that it is when it hits 'G37', e.g. : `G37* [...]

This error is showing up because between G36 and G37 (start and end of region) there were no line draws, which makes this region invalid. It is possible that line draws in this region were not correctly recognized though.

> I've uploaded a second example in diffgerber (https://github.com/ajw287/diffgerber/tree/main/examples) if you open the 'front copper' layer, you can see that most pads aren't rendered, the PCB was output from kicad.

Could you please create clean list of what images are outputs from what gerber files? Preferably with links to both images and gerber files. For now it's a bit ambigous and I'm having hard time to match them.

Argmaster, 2023-09-08:
> Just to say, pygerber is doing great things! I will show the current setup off at KiCon this weekend - I mention pygerber 1.0 and 2.0 and your name - I hope that's ok.

I'm happy to hear that, I will be great.

Argmaster, 2023-09-08:
I have managed to fix second issue with (#37), but first one is not responding to precision changes which is a bit odd. I don't yet know how to fix it and it seems unlikely that I will be able to discover it soon. For now fix for first issue is going to be included in 2.0.1 release and this issue will remain open until first part is fixed.

Argmaster, 2023-09-08:
> This demo gerber file shows another render oddity, seems like keepout isn't right. Here I'm rendering the images at 400 dpi. Sorry for all the messages - just excited about getting pygerber 2.0.0 out there in the best light :D
> 
> [image] [test_gerbers.tar.gz](https://github.com/Argmaster/pygerber/files/12487328/test_gerbers.tar.gz)

As for this case, unfortunately it will not fully work for now as macros are not yet implemented, I will try to implement them in 2.1.0

Argmaster, 2023-09-08:
> I noticed this difference in rendering between the pygerber and the same gerber viewed in KiCAD gerber viewer. I don't know if the KiCAD gerber viewer is specifically removing these artifacts or [...]

I will temporarily aid that by shrinking circle sizes by one pixel in each direction, it seems to look good afterwards:

[image]

Argmaster, 2023-09-08:
I am not sure if I should also shrink other apertures similarly to circle and if this shrinking shouldn't be applied in millimeters instead of pixels.

Argmaster, 2023-09-10:
Hi, I have discovered why `WARNING:root:Created region with no boundaries.` warnings were shown, fixed it in #55.

PR #59 (2023-11-01): Release 2.1.0
## Release 2.1.0

- Fixed #37
- Added latest Gerber spec file revision 2023.08 to documentation.
- Added command line interface for PyGerber 2D rendering.
- Added Gerber X3/X2 language server which can be acquired with
  `pip install pygerber[language-server]`. Currently server capabilities include hover
  messages with Gerber reference cited and minimal amount of suggestions. We are
  planning to further extend this server in future releases.
- Added `is-language-server-available` CLI command for checking if
  `pygerber[language-sever]` is available.
- Added support for arc regi [... cut]
closed 2023-09-12 by Argmaster