Argmaster/pygerber #103: Allocate memory error.

jekoie, 2023-12-09:
My gerber file see bellow

```console
G04 Ucamco ex. 4.6.4: Simple contour*
%FSLAX25Y25*%
%MOIN*%
%ADD10C,0.010*%
G36*
X200000Y300000D02*
G01*
X700000D01*
Y100000D01*
X1100000Y500000D01*
X700000Y900000D01*
Y700000D01*
X200000D01*
Y300000D01*
G37*
M02*
```

**code sample**
```python
from pygerber.gerberx3.api import (
      ColorScheme,
      Rasterized2DLayer,
      Rasterized2DLayerParams,
)

# Path to Gerber source file.
source_path = "./example/example_simple_contour.gbr"

Rasterized2DLayer(
      options=Rasterized2DLayerParams(
            source_path=source_path,
            colors=ColorScheme.COPPER_ALPHA,
      ),
).render().save("output.png")
```

**console error**
```console
WARNING:root:Detected use of imperial units. Using metric units is recommended. Imperial units will be deprecated in future. (See 4.2.1 in Gerber Layer Format Specification)
Traceback (most recent call last):
  File "c:/Users/jekoi/Desktop/t/main.py", line 10, in <module>
    Rasterized2DLayer(
  File "c:\Users\jekoi\Desktop\t\pygerber\gerberx3\api\_layers.py", line 173, in render
    result_handle = draw_commands.draw()
  File "c:\Users\jekoi\Desktop\t\pygerber\backend\abstract\draw_commands_handle.py", line 21, in draw
    return self.backend.draw(self.draw_commands)
  File "c:\Users\jekoi\Desktop\t\pygerber\backend\abstract\backend_cls.py", line 97, in draw
    self._post_drawing_hook()
  File "c:\Users\jekoi\Desktop\t\pygerber\backend\rasterized_2d\backend_cls.py", line 165, in _post_drawing_hook
    self._replace_image_colors()
  File "c:\Users\jekoi\Desktop\t\pygerber\backend\rasterized_2d\backend_cls.py", line 181, in _replace_image_colors
    rgba_img[np_img == gray_value] = rgba
numpy.core._exceptions.MemoryError: Unable to allocate 240. MiB for an array with shape (63007000,) and data type int32     
PS C:\Users\jekoi\Desktop\t> 
```

Another, I notice the [pcb-tools][1] nobody maintained, and you re-new a new gerber project.I use part of [pcb-tools][1] projects to draw PyQt/PySide graph.But I don't now how to use your code draw PyQt/PySdie shape.I had read small part of your code,but I not found the code  that draw the shape(like [cairo][2]).

 [1]:  https://github.com/curtacircuitos/pcb-tools
[ 2]: https://pycairo.readthedocs.io/en/latest/

Argmaster, 2023-12-09:
It looks like allocation error caused by insufficient availability of resources at runtime? How much unused RAM do you have while running this code? 
However, even if it is caused by insufficient memory availability It might be a good idea to optimize color replacement memory requirements.

jekoie, 2023-12-09:
the unused memory is about 2-3G,the resource is sufficient.

Argmaster, 2023-12-11:
I will need more of your help with reproducing this issue, as I am not able to reproduce this issue within my development evrionment. Could you please provide contents of `pip freeze`, OS, CPU, RAM (size) and full Python version (`python -VV`), so I will have a better idea about your environment? 

Peak memory usage for your Gerber file is around 1.6GB at my system, therefore I am still considering this a possible cause, thus OS might have refused allocating continuous block of 240MiB

Please try using lower DPI setting and verify if issue occurs also to smaller DPIs.

Argmaster, 2023-12-13:
@jekoie please let me know if you are willing to help with reproduction.

Argmaster, 2024-01-17:
Pinging @jekoie

closed 2024-10-26 by Argmaster