Argmaster/pygerber #109: Great new function of PyGerber

Pauloretty, 2024-01-01:
Hi. I am using this library to display rasterized png layers of Gerber file on uv screen. 

I have encountered a problem: when rendering a layers of gerber file, the unoccupied part of the image is cut off. 
In my script Pillow automatically sets the image to the corner of the screen, but because the images are cropped they don't fit together. 

Is there any way to position the rendered images relative to each other?

[image]

Thank you very much!

Argmaster, 2024-01-01:
Yes, but it is not as straight forward as one might wanted it to be. What you are after is coordinate origin and bounding box of all draw commands. both of those can be for sure extracted from Backend instance, when you are directly instantiating Tokenizer, Parser and Backend classes. I am not sure if it is currently possible with high level API and it is not possible with CLI. You will still have to manually resize individual images to match biggest one and reposition them to make their coordinate origins match, but it is possible. I'll provide more details soon.

Argmaster, 2024-01-01:
I was mistaken about high level API not being enough to handle this issue, you can use RenderingResult.get_properties() to extract LayerProperties object. See example below, based on example from readme.

```python
from __future__ import annotations
from io import BytesIO  # noqa: D100
from PIL import Image

from pygerber.gerberx3.api import (
    ColorScheme,
    Rasterized2DLayer,
    Rasterized2DLayerParams,
)

source_code = """
%FSLAX26Y26*%
%MOMM*%
%ADD100R,1.5X1.0X0.5*%
%ADD200C,1.5X1.0*%
%ADD300O,1.5X1.0X0.6*%
%ADD400P,1.5X3X5.0*%
D100*
X0Y0D03*
D200*
X0Y2000000D03*
D300*
X2000000Y0D03*
D400*
X2000000Y2000000D03*
M02*
"""

result = Rasterized2DLayer(
    options=Rasterized2DLayerParams(
        source_code=source_code,
        colors=ColorScheme.SILK,
        dpi=3000,
    ),
).render()
output = BytesIO()

result.save(output, format="png")

output.seek(0)
Image.open(output).show()

p = result.get_properties()
print(p.gerber_bounding_box)
print(p.gerber_coordinate_origin)
print(p.target_bounding_box)
print(p.target_coordinate_origin)
```

Difference between gerber_* and target_* is insignificant in our case, stick to target_* ones, as they refer to actual image.
See [LayerProperties](https://argmaster.github.io/pygerber/v2.1.0/reference/pygerber/gerberx3/api/__init__.html#pygerber.gerberx3.api.LayerProperties) in docs.

Pauloretty, 2024-01-02:
Yep, a few simple calculations with LayerProperties.target_bounding_box and LayerProperties.target_coordinate_origin solved my problem. Thanks!

closed 2024-01-02 by Pauloretty