Argmaster/pygerber #167: it does not get the pads right.

tapegoji, 2024-03-24:
**Describe the bug**
I tried to use it to output an image from a gerber file, attached. However, the pads are not showing in the gerber file. the gbr file is from  Kicad demp usb stick. 

[image]

A clear and concise description of what the bug is.

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: [e.g. iOS]
- Python version: [e.g. 3.8.13]
- PyGerber version: [e.g. 2.0.0]

**Additional context**

Add any other context about the problem here.

Argmaster, 2024-03-24:
Could you please provide Gerber source file?
 You can attach it to a comment.

tapegoji, 2024-03-30:
It's is kicad demo files, that I have attended in my original post.

Argmaster, 2024-03-30:
Ok, I understand, but I don't know which file and how to find it. Please either attach it to comment or link to online resource containing Gerber source code.

tapegoji, 2024-03-30:
Please find attached the zip folder for the Gerber files. I attached them in my original post, however GitHub decided to convert it into an image. sorry about that
[USB Stick.zip](https://github.com/Argmaster/pygerber/files/14811829/USB.Stick.zip)

Argmaster, 2024-03-30:
Lovely, I will look into it soon. Thanks for reporting this issue.
Last small thing, can you tell me if you were using command line or Python API interface?

tapegoji, 2024-03-30:
I was using python API interface in a python file.

Argmaster, 2024-03-31:
Unfortunately it does not support drill files.
I will look into that after readching PyGerber 3.0

Argmaster, 2024-04-10:
Ok, from what I see it should work with next release of PyGerber with APIv2.
Currently you can check it out by installing development version of PyGerber:

```bash
pip install pygerber@git+https://github.com/Argmaster/pygerber
```

Here is a code which uses mentioned APIv2:

```python
from __future__ import annotations

from pathlib import Path

from pygerber.gerberx3.parser2.parser2 import Parser2
from pygerber.gerberx3.renderer2.raster import (
    ImageFormat,
    PixelFormat,
    RasterFormatOptions,
    RasterRenderer2,
    RasterRenderer2Hooks,
)
from pygerber.gerberx3.tokenizer.tokenizer import Tokenizer

SOURCE = Path(
    "<path-to-grb-file>",
).read_text()

def render() -> None:
    stack = Tokenizer().tokenize(SOURCE)
    cmd_buf = Parser2().parse(stack)
    ref = RasterRenderer2(RasterRenderer2Hooks(dpmm=100)).render(cmd_buf)
    ref.save_to(
        "output.png",
        RasterFormatOptions(image_format=ImageFormat.PNG, pixel_format=PixelFormat.RGB),
    )

if __name__ == "__main__":
    render()

```

While checking it out please make sure to replace `"<path-to-grb-file>"` with real local path to file :D

closed 2024-04-11 by Argmaster