Argmaster/pygerber #180: V2 API incorrect pad rotation

tgbl-mk, 2024-04-12:
V2 API is now successfully rendering rounded rectangle pads which is good. The downside seems to be that it renders some rounded rectangle pads rotated 90 degrees. See a snippet below.

[image]

gerber .png file generated with the following code(I have picked the relevant parts from a larger script, let me know if you need more context):

```python
from pygerber.gerberx3.api.v2 import ColorScheme, GerberFile, PixelFormatEnum
from pygerber.common.rgba import RGBA

color_scheme = ColorScheme(
    background_color=RGBA.from_hex("#000000FF"),
    clear_color=RGBA.from_hex("#000000FF"),
    solid_color=RGBA.from_hex("#FFFF00"),
    clear_region_color=RGBA.from_hex("#000000FF"),
    solid_region_color=RGBA.from_hex("#FFFF00"),
)

GerberFile.from_file(
    file,
).parse().render_raster(
    gerber_png_path,
    color_scheme=color_scheme,
    dpmm=20,
    pixel_format=PixelFormatEnum.RGBA,
)
```
from the attached gerber file [PT-E03541.zip](https://github.com/Argmaster/pygerber/files/14957106/PT-E03541.zip)

- OS: Windows 10
- Python version: 3.11.0
- PyGerber version: 2.3.0
- Gerber Generated with Altium 23.5.1

This happens on multiple gerber files so I can provide more examples if needed.

Argmaster, 2024-04-12:
Thanks for reporting this issue! I will look into it soon :D

Argmaster, 2024-04-13:
Hey, I have provided a workaround targetting rotation of primitive code 21 center line which was incorrectly rendered in your files, it is available on latest development version, you can check it out by installing PyGerber directly from repository:

```
pip install git+https://github.com/Argmaster/pygerber
```

I have decided that for now I won't spend time on implementing rotation for everyting primitive which according to standard should be rotateable. I will wait for possible bug reports and add missing features when they are requested, as I do not have sample files to test it myself.

<p align="center">
  [image]
</p>

tgbl-mk, 2024-04-15:
Hey thanks for the quick turnaround.  Everything is rendering correctly now!

closed 2024-04-15 by tgbl-mk