Argmaster/pygerber #203: Incorrect Board Outline Rendering

tgbl-mk, 2024-05-28:
**Describe the bug**

When rendering board outlines with rounded edges pygerber is sometimes generating an inverted arc.
[image]

**To Reproduce**
I have included all the files required to reproduce the error in this zip.
[board_outline_error.zip](https://github.com/Argmaster/pygerber/files/15469569/board_outline_error.zip)
Including the below python script.
``` python
from pathlib import Path

from pygerber.common.rgba import RGBA
from pygerber.gerberx3.api.v2 import ColorScheme, GerberFile, PixelFormatEnum, Project, COLOR_MAP_T, FileTypeEnum

DEFAULT_COLORSCHEME = ColorScheme(
    background_color=RGBA.from_hex("#000000CF"),
    clear_color=RGBA.from_hex("#000000CF"),
    solid_color=RGBA.from_hex("#CFCF00"),
    clear_region_color=RGBA.from_hex("#000000CF"),
    solid_region_color=RGBA.from_hex("#CFCF00"),
)

DEFAULT_COLORMAP: COLOR_MAP_T = {FileTypeEnum.OTHER: DEFAULT_COLORSCHEME}

outline_file = Path("./outline.gbr")
gerber_file = Path("./bottom_mask.gbr")

def _create_gerber_png(board_outline_file: str , file: str, type: str, output_path: Path) -> None:

    gerber_png_path: Path = output_path / Path(f"{type}.png")
    Project(
        [
            GerberFile.from_file(board_outline_file, FileTypeEnum.OTHER),
            GerberFile.from_file(file, FileTypeEnum.OTHER),
        ]
    ).parse().render_raster(
        gerber_png_path,
        color_map=DEFAULT_COLORMAP,
        dpmm=20,
        pixel_format=PixelFormatEnum.RGBA,
    )

_create_gerber_png(outline_file, gerber_file, "test", "./")
```

**Expected behavior**

Expect the board outline to show the rounded edges in the correct orientation, as shown below (altium gerber viewer of the attached board outline file)
[image]

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: windows 10
- Python version: 3.11.0
- PyGerber version: 2.3.1

**Additional context**

Gerbers generated from Altium 23.5.1. I have other designs which render some corners correctly, so I suspect this has something to do with how the rounded edge is created in altium affecting the gerber output.
[image]

If you need any additional information please let me know. Thanks!

Argmaster, 2024-05-28:
Hi! Thanks for reporting this issue. I bet that here is something wrong with how arc orientation is calculated. Arcs are true pain in the ass all the time, I will look into it soon. You can check how it is rendered in SVG mode for reference.

Argmaster, 2024-05-28:
One question - can I permanently include attached Gerber files in PyGerber test suite (and hence this repository)? I would include a README file next to those files with mentioning of author and possibly some additional text you feel like should be included.

tgbl-mk, 2024-05-28:
That's a good point, I haven't checked how this behaves with SVG output. 

Feel free to include the gerbers in the test suite, no need add any author info/credit.

Argmaster, 2024-05-29:
Hi! Seems like I was able to fix the issue. You can check it out before it is deployed as part of PyGerber 2.3.2 by installing development version directly from fix branch:

```
pip install git+https://github.com/Argmaster/pygerber@fix/203
```

tgbl-mk, 2024-05-30:
Hey, 

Checked with 3 designs that were having arc rendering issues and all look as expected now. 

Thanks for the quick response!

PR #205 (2024-05-30): Fix clockwise arc rendering in raster renderer
This pull request adds exhaustive collection of arc samples and extended arc testing suite for Gerber raster renderer. Thanks to this test suite it was possible to identify a cause of #203 and fix it.

Resolves #203
commit 9bc095be (2024-05-30): Fix clockwise arc rendering in raster renderer (#205)

This pull request adds exhaustive collection of arc samples and extended
arc testing suite for Gerber raster renderer. Thanks to this test suite
it was possible to identify a cause of #203 and fix it.

Resolves #203
closed 2024-05-30 by tgbl-mk