Argmaster/pygerber #242: Wrong orientation of apertures when rendering to svg

marekpechal, 2024-07-05:
**Describe the bug**

When rendering a simple gerber with rectangular pads as svg, the orientation of the pads is wrong.

**To Reproduce**

Steps to reproduce the behavior:

1. Run the following with [this](https://github.com/user-attachments/files/16112235/example.zip) `example.gbr` file
`from pygerber.gerberx3.api.v2 import GerberFile`
`gf = GerberFile.from_file('example.gbr').parse()`
`gf.render_svg("output.svg")`

2. You will get the output below
[image]

**Expected behavior**
The gerber file (here opened in the gerber viewer bundled with KiCad 8.0) should actually look like this:
[image]

**Desktop:**

- OS: Windows 10
- Python version: 3.12.4
- PyGerber version: 2.4.0

**Additional context**

I suspect this is due to gerber specifying rotation angles counterclockwise while svg uses a clockwise angle convention. I managed to fix the issue in this particular case by adding a minus sign [here](https://github.com/Argmaster/pygerber/blob/722d99501cd6c7a16cec3025256a6ab3907e0c80/src/pygerber/gerberx3/renderer2/svg.py#L553). There may be similar issues with other graphical primitives which I have not checked.

Argmaster, 2024-07-06:
Thanks for reporting the issue and possible solution!
I guess I will have to hand craft suite for testing rotations to make sure what and when works.
I'll look into it soon, we will include the fix in `2.4.1`.

Argmaster, 2024-07-06:
One more question: Can I inculde file you have shared in PyGerber repository `test/assets/gerberx3/issues/242` for testing purposes? I can include some copyright notice next to it if that helps.

marekpechal, 2024-07-06:
Glad I could help and of course, feel free to include the file I attached
(no need for a copyright notice).

Just in case you're interested why I'm looking at pygerber - I'm currently
developing some code to generate pcb designs for our research group. We're
not writing gerbers directly - just generating kicad projects; but I quite
like pygerber as a way to convert the final gerbers into some easier to
work with format (like svg), which we could then use for automatic
validation. If we do end up using pygerber, I would be happy to report on
any issues that pop up.

Thanks for the nice work!

Argmaster, 2024-07-06:
Thank you for allowing us to use the file you attached. 
I'm glad to hear about your interest in pygerber and its potential use in your research group's PCB design projects.

We're always keen to hear how our tools are being used and would appreciate any feedback or reports on issues you encounter if you do decide to use pygerber.

We wish you the best of luck in finding the right tool for your needs, whether it's pygerber or another solution. Thank you for the kind words and for considering our work.

PR #243 (2024-07-06): Fix incorrect rotation of rectangle flashes
Changelog:
- Fixed rotation of center line (21) primitive in SVG renders.
- Fixed rendering of rotated center line (21) primitive in PNG renders.
- Fixed calculation of bounding box rotations for rectangle flashes.
- Added test cases for center line (21) primitive with different rotations.

Resolves: #242
commit b145abbb (2024-07-06): Fix incorrect rotation of rectangle flashes (#243)

Changelog:
- Fixed rotation of center line (21) primitive in SVG renders.
- Fixed rendering of rotated center line (21) primitive in PNG renders.
- Fixed calculation of bounding box rotations for rectangle flashes.
- Added test cases for center line (21) primitive with different
rotations.

Resolves: #242
commit f34c0ed6 (2024-08-11): Fix incorrect rotation of rectangle flashes (#243)

Changelog:
- Fixed rotation of center line (21) primitive in SVG renders.
- Fixed rendering of rotated center line (21) primitive in PNG renders.
- Fixed calculation of bounding box rotations for rectangle flashes.
- Added test cases for center line (21) primitive with different
rotations.

Resolves: #242
closed 2024-07-06 by Argmaster