Argmaster/pygerber #303: pyparsing.exceptions.ParseException

Feiyuyu0503, 2024-09-24:
**Describe the bug**
pyparsing.exceptions.ParseException: Expected {{End of file end of expression} | {Program stop end of expression}}, found 'I0J26016'  (at char 1312652), (line:5045, col:1222912)

**To Reproduce**
```
file = 'xxx.gbr'

def main():

    GerberFile.from_file(
        file_path=file,
    ).parse(on_parser_error=OnParserErrorEnum.Ignore).render_raster(
        f"{file[:-4]}.png",
        #dpmm=105,
        #color_scheme=ColorScheme.COPPER_ALPHA,
        #pixel_format=PixelFormatEnum.RGBA,
    )
```
**Expected behavior**
render correct

**Desktop (please complete the following information):**

- OS: ubuntu 20.04
- Python version: Python 3.12.4
- PyGerber version: pygerber, version 3.0.0-alpha

[gbr.zip](https://github.com/user-attachments/files/17107389/gbr.zip)

Argmaster, 2024-09-24:
I will look into it, funky syntax though. If possible, fix will be included in 3.0.0a2, watch out for changes in API design.

Argmaster, 2024-09-24:
Could you please provide reference image showing what is the expected result? It could be for example a screenshot from software you have used for generating this file.

Argmaster, 2024-09-24:
[image]

Is it something like this?

Feiyuyu0503, 2024-09-25:
> ![xxx](https://private-user-images.githubusercontent.com/56170852/370367112-8bd1066a-e89f-4b02-a62c-c102375e0edb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyMjYxMTIsIm5iZiI6MTcyNzIyNTgxMiwicGF0aCI6Ii81NjE3MDg1Mi8zNzAzNjcxMTItOGJkMTA2NmEtZTg5Zi00YjAyLWE2MmMtYzEwMjM3NWUwZWRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI1VDAwNTY1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQyMWY2YWI4NDRiMTIzMTY2MTEwMDRhMGE2Nzk4ZjUyMmZhYWI3ZWU3NTM1Njg0MmM5M2VkYWM3MDZhYjQ2MDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-T92OkkFkMVtBSqriOadLm6vLTLnzaZAChNLPTVcxQg)
> 
> Is it something like this?

Yes！Thank you for your help！

Feiyuyu0503, 2024-09-25:
> ![xxx](https://private-user-images.githubusercontent.com/56170852/370367112-8bd1066a-e89f-4b02-a62c-c102375e0edb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3Lm [...]

Could you please edit this reply and remove the reference image? 🙏

closed 2024-09-27 by Feiyuyu0503