Argmaster/pygerber #321: [PyGerber]: Pre-Release 3.0.0a3 Example code exported as SVG, SVG fails to display properly

chenzanyu, 2024-10-17:
win10 system, I have installed the Shapely library
my code 
```
from pygerber.gerber.api import GerberFile, Style

from pygerber.examples import ExamplesEnum, get_example_path

path_to_gerber_file = get_example_path(ExamplesEnum.UCAMCO_2_11_2)

image = GerberFile.from_file(path_to_gerber_file).render_with_shapely(
    Style.presets.BLACK_WHITE
)
image.save("output.svg")
```

[image]

Argmaster, 2024-10-18:
Hi, for me it seems like I see an image in the issue, maybe it is a bit small, but still visible:

[image]

Could you elaborate on what software are you using to view the image?
Could you please make sure that this is not caused by the image being same color as the background by using different color preset? SVG images do not have background, so there is only white image being created.

Argmaster, 2024-10-26:
@chenzanyu hi, could you please follow up on that?

chenzanyu, 2024-10-26:
Apologies, I was delayed by some matters.   You are right, the issue was with '**Style.presets.BLACK_WHITE**'. The fill color is white, which caused it to not display on a white background. I tried changing it to '**Style.presets.COPPER**', and the SVG displays correctly now. I think we can close this issue.

Argmaster, 2024-10-26:
Thanks for the response, I'm happy I could help, have a good day!

closed 2024-10-26 by Argmaster