Argmaster/pygerber #322: [PyGerber]: numpy 2 support

BlueDrink9, 2024-10-19:
# PyGerber Bug Report

I need both pygerber and numpy >2 for a project. Is there anything holding pygerber back from numpy v2?

## Mandatory checks

### Before opening this issue:

- [x] I have reviewed the README for guidelines and haven't found a solution there.
- [x] I have reviewed the PyGerber documentation and haven't found a solution there.
- [x] I have reviewed the existing open issues and verified that this is not a
      duplicate.
- [x] I have reviewed the existing closed issues and verified that this was already
      resolved or marked as won't fix.
- [x] I have reviewed the existing pull requests and verified that this is not a already
      known issue.
- [x] I have reviewed the existing discussions and verified that this is not a already
      known issue.

## To Reproduce

Create a `requirements.txt`containing:

```
pygerber > 2
build123d @ git+https://github.com/gumyr/build123d.git
```

`pip install -r requirements.txt`

This will fail to resolve dependencies because pygerber requires numpy < 2, but build123d has just updated to using numpy >2.

## Expected behavior

Successful installation

## Additional context

> If applicable, add screenshots, code examples, or any other resources that can speed
> up process of reproducing and fixing the issue.

> If your issue was discovered with use of specific source file (e.g. in Gerber format),
> please attach it to the issue. If file is confidential, please create minimal subset
> of the file which can be shared in public and allows to reproduce the issue.

> If you are not able to create minimal reproduction for confidential source, you can
> email me the confidential file at `argmaster.world@gmail.com` with the issue title in
> the subject. I will not share the file anywhere in public and will delete it after the
> issue is resolved. We can discuss NDA agreements via email if needed.

> You still should create a public issue based on this template for the issue to be
> investigated, please use `<confidential-file>`, `<confidential-image>`, etc. as a
> placeholders for the confidential files.

## Environment:

**Please complete the following information:**

- Python version: 3.11
- PyGerber version: any

Argmaster, 2024-10-19:
Hi,
numpy dependency is tricky, if you can figure out configuration of numpy versions for PyGerber in a way that keeps it running both on 3.8 and 3.12, I'll be happy to accept it. In PyGerber numpy usage is very limited and unless there were changes in core functionality like ndarray mul/add etc. it should be fine. Also, likely Pillow has it's own expectations to numpy version, so it may also be neccessary to apply come constraints on its version.

Right now we have this:

```toml
numpy = [
    { version = "^1.24.4", markers = "python_version >= '3.8' and python_version <= '3.11'" },
    { version = "^1.26.1", markers = "python_version >= '3.12' and python_version < '3.13'" },
]
```

Because you can't use `numpy>=1.24` on Python 3.8. I don't know what is a minimal required Python version for numpy 2.x, but we could basically change the ranges to be something like:

```toml
numpy = [
    { version = "^1.24", markers = "python_version >= '3.8' and python_version <= '<numpy-2-min-py>'" },
    { version = "<3", markers = "python_version >= '<numpy-2-min-py> and python_version < '3.13'" },
]
```

So we don't break environments of people running on 1.26

Argmaster, 2024-10-19:
Actually, it is possible that changing `^1.26.1` to `>=1.26,<3` would do the trick.

BlueDrink9, 2024-10-28:
Looking into this, it's not clear that you even need to specify a numpy version. As far as I can tell, all you use it for is [creating a basic ndarray for shapely's transform](https://github.com/Argmaster/pygerber/blob/b461008949c22b75419ae00f7373cceed0226cf0/src/pygerber/vm/shapely/vm.py#L228) - I can only see these two uses.

In light off that, I set numpy version specification to just `numpy = ">=1.24.4,<3"` and pygerber installed with dependencies without issue. Are you happy to make that change?

Argmaster, 2024-10-28:
Thanks for looking into this. Could you please open a pull request for that?

Argmaster, 2024-11-09:
Hi,

I have relaxed version requirements for development version of PyGerber 3.0.0a4, right now on `main`.

I will be back porting what is possbile for `2.4.x` and `2.3.x`.

BlueDrink9, 2024-11-10:
Thank you! Sorry for not opening the PR earlier

Argmaster, 2024-11-11:
Hi @BlueDrink9, could you please check if PyGerber 2.4.2 works for you?

closed 2024-12-24 by Argmaster

BlueDrink9, 2025-02-27:
Sorry for not following upon this. Just chiming into say that yes, this worked fine for my purposes.