Argmaster/pygerber #333: [PyGerber]: When I am trying to import and render .art Gerber Files from Cadence Allegro, it shows error

Rajkisan, 2024-11-06:
# PyGerber Bug Report

## Mandatory checks

> Checks listed below are mandatory for opening a new issue.

> If haven't done any of the checks listed below, please do that, we will not look into
> the issue until all of the checks in mandatory checks section are checked.

### Before opening this issue:

- [✓] I have reviewed the README for guidelines and haven't found a solution there.
- [✓] I have reviewed the PyGerber documentation and haven't found a solution there.
- [✓] I have reviewed the existing open issues and verified that this is not a
      duplicate.
- [✓] I have reviewed the existing closed issues and verified that this was already
      resolved or marked as won't fix.
- [✓] I have reviewed the existing pull requests and verified that this is not a already
      known issue.
- [✓] I have reviewed the existing discussions and verified that this is not a already
      known issue.

## This is my code

from pygerber.gerber.api import GerberFile

# Function to read the file content and process it with pygerber
def process_gerber_file(file_path):
        # Open the Gerber file and read the entire content into a string
        with open(file_path, 'r') as gerber_file:
            gerber_content = gerber_file.read()

        # Create a GerberFile object using pygerber and load the content as a string
        outline_layer = GerberFile.from_str(gerber_content)
        vm = outline_layer.render_with_pillow()
        bounding_box = vm.get_image_space()
        width = bounding_box.max_x_mm - bounding_box.min_x_mm
        height = bounding_box.max_y_mm - bounding_box.min_y_mm
        print(width, height)

# Example usage: Replace with the path to your Gerber file
gerber_file_path = "BOTTOM.art"
process_gerber_file(gerber_file_path)

It shows raise EmptyAutoSizedLayerNotAllowedError(top_layer.layer_id)
pygerber.vm.types.errors.EmptyAutoSizedLayerNotAllowedError: id='%main%'

## Expected behavior

> It has to return the size by width and height of the layer

## Environment:

**Please complete the following information:**

- Operating system: [`macOS24.04`]
- Python version: [`3.12.0`]
- PyGerber version: [`3.0.0a3`]

Rajkisan, 2024-11-06:
[Flexray_board_fabrication_and_assembly.zip](https://github.com/user-attachments/files/17646282/Flexray_board_fabrication_and_assembly.zip)
TOP.art and BOTTOM.art

Rajkisan, 2025-02-27:
Any update in this ?

Argmaster, 2025-02-28:
One of the problem with that files is following line: `%IR0*IPPOS*OFA0.00000B0.00000*MIA0B0*SFA1.00000B1.00000*%` - this is not a valid Gerber code, merging multiple extended commands in one is not part of the standard. It is one of problems highlighted in #367, I have made some more comments about this problem in that issue as well as suggested a solution, but right now I don't have time to implement such heavy deviation from parser principles. If you are interested you could try following instructions from #367 comment and implement it for 3.0.0b1

As for `pygerber.vm.types.errors.EmptyAutoSizedLayerNotAllowedError`, this should go away after adding workaround for unusual extended command syntax.