Argmaster/pygerber #334: [Enhance] can provide provide pure draw shape api for third api to draw?

jekoie, 2024-11-09:
I notes that `RVMC.commands` decompose the shape to `Line` and `Arc`, just two shape.The raw gerber file have mang shapes like `Line`, `Rectangle`, `Obround`, `Polygon`, `Arc`, `Circle` and `Primitives(AM commands)`.
So can you provide basis geometry shape api, for that i can draw and manipulate the shape item by myself, not just a pictuer.For example draw gerber on `pycairo`, `Qt`, `tkinter` and so on.

lattaai5, 2024-11-09:
Hello, I tried to fix the issue

This is what I did:

Add new shape types (Rectangle, Obround, Polygon, Circle) and modify the Shape class to support these new types. Also, add a new method to Shape class for drawing on different backends.

I also created pull request: https://github.com/Argmaster/pygerber/pull/335

> [!CAUTION]
> Disclaimer: This fix was created by Latta AI and you should never merge before you check the correctness of generated code!
>
> The fix provided by Latta AI might not be complete and it can serve as an inspiration.

---
This bug was fixed for free by Latta AI - https://latta.ai/ourmission

If you no longer want Latta AI to attempt fixing issues on your repository, you can block this account.

Argmaster, 2024-11-09:
Hi,

Even though Gerber provides finite set of basi shapes, it becomes nontrivial to implement them whem we take into account transformations with different origins they have support. This would require a lot of code to handle individual shapes separately if you can even figure it all out considering how tricky composed transformations can get. With line/arc approach, everything is much more viable to implement and extend in the future, since everything is a collection of points.

With that said, If you want to create image or gerber file, please look into builder APIs, If you want to modify the code, you will probably need to wait for optimizer API to handle modification on existing files at Gerber AST level. This one is tricky, as it may appear that modifying Gerber file is as trivial as adding single draw wherever we feel like, but since Gerber is systateful, you have to always account for changes in stste you have to do and undo to achieve shape with desired properties.

If you want to implement rendering of Gerber files with different tools, like `pycairo` you have mentioned, the intended way is to create class deriving from VirtualMachine similarily to how PillowVirtualMachine works and implement drawing of shapes form series of lines and arcs. You would still need to handle it even if everything was based on Gerber shapes, as there are things like regions and outline primitives with require you to draw arbitrary shapes from series of points.

In the future I will be looking into providing AST walking with hooks for more abstract concepts like shapes etc instead of individual AST node, but research has to be done here first.

Argmaster, 2024-11-09:
> Hello, I tried to fix the issue
>
> This is what I did:
>
> Add new shape types (Rectangle, Obround, Polygon, Circle) and modify the Shape class to support these new types. Also, add a new method to Shape [...]

Useless blob of code you have made, neither it solves the issues nor it works.

Argmaster, 2024-11-11:
@jekoie I would appreciate feedback from you regarding this issue and my response 😄

jekoie, 2024-11-13:
Only use points just use fitting method(like Taylor series), those points approximate a polygon, this will made draw method cost more time and slowly.

Argmaster, 2024-11-13:
Ok, could you provide proof of concept for your request? Some basic code how would you incorporate in extisting source. Then we can follow up on that.