Argmaster/pygerber #357: Render Project as SVG

clemjvdm, 2024-12-17:
Hello, currently I believe a project can only be rendered to a rasterized image. It would be nice to be able to export it as an SVG as well.

Argmaster, 2024-12-24:
Thank you for your interest in PyGerber project. I am glad to receive enhancement suggestions. Right now I don't have capacity to implement this feature, I will look into it in the future. It is planned as part of release 3.0.0, but there is no exact timeline for said release. If you want, you can try implementing it yourself, I can provide help and general guidelines if needed. We are always happy to receive pull requests!

clemjvdm, 2024-12-24:
Thanks for your reply. I understand. I'd be happy to try and implement it. Where should I reach out for some advice? The discussion board?

Argmaster, 2024-12-24:
We can continue this discussion here, in comments. 
In general what you should probably do is to use shapely representation of image to merge multiple images (geometries) into one image. Im not sure if shapely supports concept of z-index, you would need to verify it. After composing shapely geometry into one image you would likely be able to use existing shapely -> svg export implementation or something really similar.

Have a look at implementaiton details for existing svg and raster export code here:

- https://github.com/Argmaster/pygerber/blob/main/src%2Fpygerber%2Fgerber%2Fapi%2F_composite_view.py

- https://github.com/Argmaster/pygerber/blob/main/src%2Fpygerber%2Fgerber%2Fapi%2F_gerber_file.py

- https://github.com/Argmaster/pygerber/blob/main/src%2Fpygerber%2Fvm%2Fshapely%2Fvm.py

If shapely doesnt provide z-index-like functionality, you could instead reimplement svg export and stack the geometry into one image, but this requires taking into account that different layers can have different size and offsets. Afaik svg should support z-index so at least this part should go smoothly.

To sum up, start by looking into existing svg export code and shapely vm to have general idea how we handle vector geometry and svg export, then have a look into raster exports with pillow to get an idea what we have to account for while exporting multiple images at once. Afterwards try figuring out if you can utilize shapely geometry union to merge geometry or svg format to merge multiple images into one.

Tommy0295, 2025-01-09:
Hello,
I would appreciate this feature also very much.
@clemjvdm did you get it to work or tried it at all?

clemjvdm, 2025-01-09:
I haven't worked on it yet, but I plan to soon.

Argmaster, 2025-02-01:
Hi, @clemjvdm, @Tommy0295

Did you have time to work on this? If you couldn't find time to work on it, I would prefer to take it over to try implementing this feature for 3.0.0b1. If you have worked on this, can you please submit a pull request with whatever code you have crafted so we can track progress on that?  

I appreciate your good intentions regardless.

Argmaster, 2025-06-08:
I think I didn't have time to work on it since last update in this issue, I kind of burned out when it comes to working on PyGerber. I will likely return in some time, but for now I am working on different project. Feel free to make an attempt on implementing this as we discussed previously, guidelines should be still up to date. You can find basic contribution guidelines [here](https://argmaster.github.io/pygerber/dev/90_development/40_contribution_guidelines.html)

clemjvdm, 2025-06-09:
Sorry to hear that. Enjoy working on the different project :). I'll give it a try then and see how it goes.