Argmaster/pygerber #361: [PyGerber]: When I try to render with .gbr files, getting error pyparsing.exceptions.ParseException

Rajkisan, 2025-01-12:
# PyGerber Bug Report

## Mandatory checks

> Checks listed below are mandatory for opening a new issue.

> If haven't done any of the checks listed below, please do that, we will not look into
> the issue until all of the checks in mandatory checks section are checked.

### Before opening this issue:

- [x] I have reviewed the README for guidelines and haven't found a solution there.
- [x] I have reviewed the PyGerber documentation and haven't found a solution there.
- [x] I have reviewed the existing open issues and verified that this is not a
      duplicate.
- [x] I have reviewed the existing closed issues and verified that this was already
      resolved or marked as won't fix.
- [x] I have reviewed the existing pull requests and verified that this is not a already
      known issue.
- [x] I have reviewed the existing discussions and verified that this is not a already
      known issue.

## This is my code

```python
from pygerber.examples import ExamplesEnum, get_example_path
from pygerber.gerberx3.api.v2 import ColorScheme, GerberFile, PixelFormatEnum

GerberFile.from_file(
    "GerberKicad/dse_vmj_ex_004_legend_top.gbr",
).parse().render_raster(
    "output.png",
    dpmm=100,
    color_scheme=ColorScheme.COPPER_ALPHA,
    pixel_format=PixelFormatEnum.RGBA,
)
```

It shows
```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/rajki/Downloads/Glonix/vm.py", line 6, in <module>
    ).parse().render_raster(
      ^^^^^^^
  File "/Users/rajki/Downloads/Glonix/.venv/lib/python3.12/site-packages/pygerber/gerberx3/api/_v2.py", line 197, in parse
    tokens = Tokenizer().tokenize(self.source_code)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/rajki/Downloads/Glonix/.venv/lib/python3.12/site-packages/pygerber/gerberx3/tokenizer/tokenizer.py", line 49, in tokenize
    return self._tokenize_grammar(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/rajki/Downloads/Glonix/.venv/lib/python3.12/site-packages/pygerber/gerberx3/tokenizer/tokenizer.py", line 84, in _tokenize_grammar
    ast = grammar.parse_string(source, parse_all=parse_all)[0]
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/rajki/Downloads/Glonix/.venv/lib/python3.12/site-packages/pyparsing/core.py", line 1216, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected End of file, found '%'  (at char 198), (line:4, col:1)
```

## Expected behavior

It has to render the image of the layer

## Environment:

**Please complete the following information:**

- Operating system: [`macOS24.04`]
- Python version: [`3.12.0`]
- PyGerber version: [`2.4.2`]

Rajkisan, 2025-01-12:
Contents of top.gbr file

```
G04 CAMtasticDXP RS274-X Output*
G04 Serial Number: 0000-00-00000*
G04 File Name:D:\SABHAKANT MISHRA\ALTIUM\VMJ EXCITER\NEW\DSE_VMJ_EX_004\DSE_VMJ_EX_004 GERBER DATA\dse_vmj_ex_004_copper_signal_top.gbr *
%FSDAX44Y44*%
%MOMM*%
%SFA1B1*%

%IPPOS*%
%AMD31*
4,1,8,-0.774700,-0.909320,0.774700,-0.909320,0.909320,-0.774700,0.909320,0.774700,0.774700,0.909320,-0.774700,0.909320,-0.909320,0.774700,-0.909320,-0.774700,-0.774700,-0.909320,0.0*
1,1,0.273000,-0.774700,-0.774700*
1,1,0.273000,0.774700,-0.774700*
1,1,0.273000,0.774700,0.774700*
1,1,0.273000,-0.774700,0.774700*
%
%AMD69*
4,1,4,-1.816100,0.990600,0.990600,-1.816100,1.816100,-0.990600,-0.990600,1.816100,-1.816100,0.990600,0.0*
%
%AMD70*
4,1,4,-0.236220,1.163320,-1.163320,0.236220,0.236220,-1.163320,1.163320,-0.236220,-0.236220,1.163320,0.0*
%
%AMD73*
4,1,4,-0.137160,-0.513080,0.513080,0.137160,0.137160,0.513080,-0.513080,-0.137160,-0.137160,-0.513080,0.0*
%
%AMD75*
4,1,4,-0.053340,-0.429260,0.429260,0.053340,0.053340,0.429260,-0.429260,-0.053340,-0.053340,-0.429260,0.0*
%
%AMD76*
4,1,4,-0.678180,-2.278380,2.278380,0.678180,0.678180,2.278380,-2.278380,-0.678180,-0.678180,-2.278380,0.0*
%
%AMD77*
4,1,4,0.177800,0.683260,-0.683260,-0.177800,-0.177800,-0.683260,0.683260,0.177800,0.177800,0.683260,0.0*
%
%AMD78*
4,1,4,-0.627380,0.248920,0.248920,-0.627380,0.627380,-0.248920,-0.248920,0.627380,-0.627380,0.248920,0.0*
%
%AMD79*
4,1,4,0.210820,0.746760,-0.746760,-0.210820,-0.210820,-0.746760,0.746760,0.210820,0.210820,0.746760,0.0*
%
%AMD80*
4,1,4,-0.147320,0.449580,-0.449580,0.147320,0.147320,-0.449580,0.449580,-0.147320,-0.147320,0.449580,0.0*
%
%AMD85*
4,1,4,0.000000,0.815340,-0.815340,0.000000,0.000000,-0.815340,0.815340,0.000000,0.000000,0.815340,0.0*
%
%AMD90*
4,1,4,-0.416560,0.073660,0.078740,-0.416560,0.416560,-0.073660,-0.078740,0.416560,-0.416560,0.073660,0.0*
%
%AMD91*
4,1,4,0.078740,0.416560,-0.416560,-0.073660,-0.078740,-0.416560,0.416560,0.073660,0.078740,0.416560,0.0*
%
%AMD95*
4,1,4,0.147320,0.441960,-0.441960,-0.147320,-0.147320,-0.441960,0.441960,0.147320,0.147320,0.441960,0.0*
%
%AMD96*
4,1,4,-0.474980,0.124460,0.124460,-0.474980,0.474980,-0.124460,-0.124460,0.474980,-0.474980,0.124460,0.0*
%
%AMD97*
4,1,4,0.045720,0.533400,-0.533400,-0.045720,-0.045720,-0.533400,0.533400,0.045720,0.045720,0.533400,0.0*
%
%AMD100*
4,1,4,1.823720,0.749300,0.749300,1.823720,-1.823720,-0.749300,-0.749300,-1.823720,1.823720,0.749300,0.0*
%
%AMD105*
4,1,4,-0.436880,-1.104900,1.104900,0.436880,0.436880,1.104900,-1.104900,-0.436880,-0.436880,-1.104900,0.0*
%
%AMD106*
[... 2610 lines cut ...]
X00087920Y00454341D03*
G54D309*
X00537790Y00456729D03*
X00341142Y00197206D03*
X00244421Y00196915D03*
X00341472Y00292976D03*
X00243100Y00292316D03*
X00149800Y00334159D03*
X00537790Y00346947D03*
X00149800Y00283671D03*
X00364730Y00388747D03*
X00364906Y00348308D03*
G54D310*
X00682936Y00246207D03*
X00689542Y00246207D03*
X00676379Y00233076D03*
X00689542Y00232995D03*
X00682936Y00232995D03*
X00676329Y00239601D03*
X00689542Y00239601D03*
X00676329Y00246207D03*
X00682936Y00239601D03*
M02*

```

lily-PHP, 2025-01-14:
I got same error message when I parse a GBS file. 
here is my code:
`from pygerber.gerberx3.api.v2 import FileTypeEnum, GerberFile
parsed_file = GerberFile.from_file(
    "./80600_50_6-ISS1/80600_50_6-ISS1_bott_resist.GBS",
    file_type=FileTypeEnum.INFER_FROM_ATTRIBUTES
).parse()
print(parsed_file.get_info())
print(parsed_file.get_file_type())`

here I attach my gerber zip file:
[80600_50_6-ISS1.zip](https://github.com/user-attachments/files/18408499/80600_50_6-ISS1.zip)

please advise how to solve it. 
Thanks,
Lily
[80600_50_6-ISS1.zip](https://github.com/user-attachments/files/18408518/80600_50_6-ISS1.zip)

Argmaster, 2025-01-20:
@Rajkisan the problem in with your file is cause by incorrect zero coding mode. `D` is not an accepted value accoring to Gerber standard. Do you know how this zero coding mode should be interpreted?
```
%FSDAX44Y44*%
```

Rajkisan, 2025-01-20:
What Can I do now?